mcci-catena/Arduino_Core_STM32 #28: Document how to use DFU

terrillmoore, 2018-08-02:
DFU is now supported. Need to pull in the instructions from pull request #24.

Split driver description up based on system in use:

1. Windows (as given in #24)
2. macOS (nothing to be done)
3. Linux (need to install the rules from `drivers/linux/mccicatena4551.rules`, restart the dbus daemon, and make sure current user is in plugdev group) -- perhaps we should grab the script from MCCI's `usbiotest`.

Ideally, we'd have a YouTube video for each system (WIndows, macOS, Linux).

vpowar, 2018-08-28:
With regard to our slack conversation, I have documented some steps I followed for uploading code to 4551 Catena using DFU. It can be found here: 
https://github.com/vpowar/LoRaWAN_SensorNetworks-Catena#uploading-code

terrillmoore, 2018-08-29:
Very nice, thank you!

svelmurugan92, 2019-04-02:
Document available here [](https://github.com/mcci-catena/HW-Designs/blob/master/Boards/Catena-4611_4612/234001173a_(Catena-4612-User-Manual).pdf)

terrillmoore, 2019-06-18:
Before I can close this, this needs to be linked from the BSP, because it's a common problem. We have a README.md, we should consider putting it there. Reassigning to @svelmurugan92 and @dhineshkumarmcci because you're working on the new BSP features.

terrillmoore, 2019-06-23:
@svelmurugan92 @dhineshkumarmcci  this is really important; I need to be able to point users to a procedure for the documeens for Windows.

> 
> 
> Document available here [https://github.com/mcci-catena/HW-Designs/blob/master/Boards/Catena-4611_4612/234001173a_(Catena-4612-User-Manual).pdf](url)

@svelmurugan, this link is broken.

svelmurugan92, 2019-06-23:
@terrillmoore Below links works. We will add this in README.md
[234001173a_(Catena-4612-User-Manual).pdf](https://github.com/mcci-catena/HW-Designs/blob/master/Boards/Catena-4611_4612/234001173a_(Catena-4612-User-Manual).pdf)