mcci-catena/Arduino_Core_STM32 #29: Clarify use of USB and regular serial ports

terrillmoore, 2018-08-02:
In v1.1.0 and earlier, if you download a test sketch and use DFU to upload it, you may or may not get serial output.

- if you have set "Serial ports: No serial", then serial output will appear on USB.
- if you have set "Serial Ports: (any other value)", then no output will appear on USB.

This is confusing to people used to the Feather M0. It happens because Serial points to the UART if available. For people unfamiliar with the board and the settings, it makes them think that the board is not working.

We should change the rules to match the Feather M0. Serial should point to UsbSerial if available, and to the first UART otherwise.

chwon64, 2018-09-13:
I think we need to remove "USB interface" menu and add "USB serial" in the "Serial interface" menu.

So "Serial interface" menu has submenu as below:
1. USB Serial (default)
2. Generic Serial

terrillmoore, 2018-09-25:
Closed by #44

closed 2018-09-25 by terrillmoore