mcci-catena/Arduino_Core_STM32 #41: Pullup by default on SPI bus looks wrong

terrillmoore, 2018-08-18:
If we skip the part of initialization that sets up the various peripherals after shutting down the BME280 and the light sensor,, the 4612 has a static current draw of 45 uA at 2.5V. Interestingly, this looks like an internal pullup resistor fighting against a ground. Check all the pullup configurations; especially in variant.cpp, many of which specify PULLUP.

chwon64, 2018-12-04:
I changed PULLUP setting for all SPI pins (MOSI, MISO, SCLK and SSEL) from PULLUP enable to no PULLUP. 
I got 30 uA at 2.5V when deep sleep mode.

terrillmoore, 2018-12-04:
That's great! We're getting close to the Grasshopper (which gets < 4 uA). Now, we have lots on the board, so that may not be possible. 

* is this the 4610 or 4612?
* What clock are you using during standby?

30uA is 385 days or so on a 1000 mAh lithium. But if we can get to 3uA, that's ten years... of course we have to transmit, too, but this is getting better and better!