mcci-catena/Arduino_Core_STM32 #66: Pullup in peripheral pins seems to be wrong

dhineshkumarmcci, 2019-01-17:
Pullup configuration for SPI pins and USART pins seems to be wrong in "PeripheralPins.c" for 4801. They should be configured as No_Pullup. #41

terrillmoore, 2019-01-17:
@dhineshkumarmcci can you modify on your side and test? I'm not sure ChaeHee is set up with a 4801 yet. Then you can submit a pull request. Thanks!

dhineshkumarmcci, 2019-01-17:
We have tested it in our end and that resulted in fall of power consumption by 210 uA.

- With Pullups - 263 uA
- With No_Pullups - 53 uA

terrillmoore, 2019-01-17:
That's great. So my suggestion is that you submit a pull request.... that way it's easy to review and we can just accept the change and move forward. In the pull request, please include the text "Fix #66:".....

commit 50d827f0 (2019-01-18): Merge pull request #67 from mcci-catena/issue66

Fix #66: Change SPI/USART pins in Catena 4801 from pull-up to no pull-up
closed 2019-01-18 by chwon64