mcci-catena/Arduino_Core_STM32 #70: Need USB + HW serial configuration in menu.

terrillmoore, 2019-02-20:
Looks like in fixing #29 we broke an important use case: USB connection combined with local serial port (as Serial1). It probably is possible to work around this in the sketch, but we ought to support this as an option.

AlexSneedMiller, 2019-06-10:
Until this is fixed, does this mean that you cannot have a serial port? Or is it just that if you have the USB cable plugged in then the board will not be able to have a serial port on different pins?

terrillmoore, 2019-06-10:
It's reltively easy to hack this in your boards.txt -- it's just the combination of #defines. In fact, the new 4630 board has serial (for the PM2.5 sensor) and USB. if you look (for example) the 4612, you'll see:

```
mcci_catena_4612.menu.xserial.usb.build.xSerial={build.usb_flags} -DNO_HWSERIAL
```

If you change that to:

```
mcci_catena_4612.menu.xserial.usb.build.xSerial={build.usb_flags} -DFIRST_THIRD_HWSERIAL
```

Then you'll get both usbserial and UARTS 1 and 3 enabled.  There may be some confusion about which port `Serial` refers to (as opposed to `UsbSerial` and `Serial1`), but otherwise things should work. In an older version, they did work.

terrillmoore, 2019-07-03:
Fixed by #98.

closed 2019-07-03 by terrillmoore