mcci-catena/Arduino_Core_STM32 #73: STM32 Low power consumption with Standby mode

dhineshkumarmcci, 2019-04-02:
In Standby mode, on enabling the Power interface clock, Low power voltage regulator, and Ultra-Low-power mode, power consumption can be reduced.

closed 2019-04-03 by chwon64

terrillmoore, 2019-04-14:
This change is incorrect. The *caller* (according to docs) is responsible for doing the power management. I have reverted this change.

terrillmoore, 2019-04-14:
In particular, there's a fundamental misunderstanding of how the WUF bit is supposed to be used.

You must (at the application level, or at least at a higher level than this)

1. clear the WUF *before* setting up the event sources.
2. set up the event sources (the RTC)
3. call `HAL_PWR_EnterSTANDBYMode()`.  

Then, in enter standby mode, if an event has occurred between 1 and 3, it wakes up immediately. If you clear WUF in`HAL_PWR_EnterSTANDBYMode()`, then you'll miss an event. This is forbidden design style according to MCCI rules; even if you don't think the event could possibly happen, we are required to code defensively.

I believe `HAL_PWR_EnterSTANDBYMode()` should clear the SBF. I've opened #80 for this.

commit 9e5d3b70 (2019-04-03): Fix #73: Enable HAL_RCC_PWR_CLK, PWR_LOWPOWERREGULATOR_ON, PWR_CR_ULP
commit c6ec2172 (2019-04-03): Merge pull request #74 from mcci-catena/issue73

Fix #73: Enable HAL_RCC_PWR_CLK, PWR_LOWPOWERREGULATOR_ON, PWR_CR_ULP
commit 1a4466e0 (2019-04-15): Revert "Fix #73: Enable HAL_RCC_PWR_CLK, PWR_LOWPOWERREGULATOR_ON, PWR_CR_ULP"

This reverts commit 9e5d3b708d7fef1e5eb7498fdd3e6514843a28fe. See
Terry's comments in #76 review for why. It needs reconsideration.