mcci-catena/Arduino_Core_STM32 #76: STM32 Low power consumption with StopWithLowPowerRegulator mode

dhineshkumarmcci, 2019-04-08:
As per STM32l0 datasheet, in Standby mode, the device used to reinitialize during wake up from sleep.
> After waking up from Standby mode, program execution restarts in the same way as after a
> Reset (boot pins sampling, vector reset is fetched, etc.).

Due to this reason, it is better to configure STM32 sleep in **StopWithLowPowerRegulator**.

Enable Power interface clock, Low power voltage regulator, and Ultra-Low-power mode under **HAL_PWR_EnterSTOPMode** in **stm32l0xx_hal_pwr.c**.

chwon64, 2019-04-11:
I don't understand exactly why standby mode is not good for us.

terrillmoore, 2019-04-11:
The system reboots on wakeup. There is no state preserved in SRAM. This is not a "sleep" api. 

I don't mind if we provide a new "hibernate" API that reboots the system on wakeup. But it can't be called "sleep". And we need an api that just sleeps, as we have many apps that don't justify the complexity of dealing with a reboot.

dhineshkumarmcci, 2019-04-11:
Initially, on using Standby mode, we do not see the reboot issue and the power is same during Standby and Stop modes because the Power Interface Clock is not enabled. To access the Power Control register, it is necessary to enable the Power Interface Clock.

On enabling the Power Interface Clock, the controller behaves as mentioned in the datasheet. That is during Standby mode, the system reboots on wake-up and we can see the difference in power for Standby mode and Stop modes.

terrillmoore, 2019-04-15:
This issue is addressed by a number of other fixes. It is true that stop mode is better, but the (old) STANDBY API must not actually (ever) enter STANDBY mode, as it will cause apps built with an older version of the Catena-Arduino-Platform to reboot. A new API is introduced by #85, that will really stand by.

commit f0f7f128 (2019-04-09): Fix #76: Remove the change with PWR_LOWPOWERREGULATOR_ON
closed 2019-04-15 by terrillmoore