mcci-catena/Arduino_Core_STM32 #94: Remove the board build old flag in boards.json

dhineshkumarmcci, 2019-06-17:
As Catena 4630 is added after the implementation of new naming`(build.board=MCCI_{something})` methodology, it do not require the backward compatibility. So I hope we can remove `build_board_old_flags` from **boards.json**.
https://github.com/mcci-catena/Arduino_Core_STM32/blob/9864576a4cc77b25efe446da907acd7ba5c5e793/boards.json#L54
@terrillmoore @chwon64 Is that ok to remove the `build_board_old_flags`?

terrillmoore, 2019-06-18:
You can remove it for new boards. *But* -- we have to document this properly in the appropriate README.md (there's an update in #96, so I reckon we have some place to document this).

dhineshkumarmcci, 2019-06-26:
Removed the `build_board_old_flags` and pushed to branch **issue95**. Added up with pull request #98

commit bc1d2dd3 (2019-06-26): Fix #94: removed build_board_old_flags for catena 4630
closed 2019-07-03 by terrillmoore