mcci-catena/Arduino_Core_STM32 #100: Add 2-hw-serial configuration to 461x boards.

terrillmoore, 2019-06-20:
It would be convenient to support two UARTs on 461x boards as well.

> Small correction: Only in 4630 and 4801 (To support PMS7003 and RS485 transceiver respectively) we enabled extra HW serial USART2 . In the 461x and 4551 we enabled only one HW serial USART1

_Originally posted by @svelmurugan92 in https://github.com/mcci-catena/Arduino_Core_STM32/pull/98_

svelmurugan92, 2019-06-24:
In 461x boards, USART2 is not available. We used those pins for VBAT and VBUS monitoring. 
I think, At this point it is difficult to change that because of backward compatibility problems. 
In 461x boards, We can use LPUART1 as second HW serial port. Those pins are PA14 and PA13. It is shared with SWCLK and SWDIO.

terrillmoore, 2019-06-24:
That would be fine. LPUART1 is very similar from a driver point of view. But no time to do it in this release. so we'll have to push this to later.