mcci-catena/Arduino_Core_STM32 #188: Make stm32_interrupt_enable() and stm32_interrupt_disable() callable from C

terrillmoore, 2021-12-12:
It appears that the intent was that `stm32_interrupt_enable()` and stm32_interrupt_disable()` should be callable from C. But the header file (cores/arduino/stm32/interrupt.h) neglected to declare them `extern "C"`, so they can't be.  (There's an overload from `stm32_interrupt_enable()` using a `std::function<>` wrapper, and the header file makes that invisible when included in a `.c` file. But since the visible `stm32_interrupt_enable()` is implemented in a C++ file, you get a link error when you try to link.  Ditto for `stm32_interrupt_disable()`.

The solution is to wrap in the C-ish things in `extern "C"` in `interrupt.h` when compiling in C++; that way, the functions will be callable from C.

dhineshkumarmcci, 2021-12-21:
@terrillmoore have tested with these changes. application works good but that requires changes in Sigfox library where interrupt_enable function is being called.

https://github.com/mcci-catena/MCCI_Sigfox_Image/blob/5f794cb04a1e47553b5d88410813491d41fde0ef/src/arduino_wrapper.c#L518-L526

I will do the necessary changes and push them to the Sigfox lib.

Also for PR #187 I will include the changes of branch `issue188`.

terrillmoore, 2021-12-21:
> requires changes in Sigfox library where interrupt_enable function is being called.

That's as expected. Thanks!