mcci-catena/Arduino_Core_STM32 #206: 4917 is not stable using 46xx bottloader

dhineshkumarmcci, 2023-08-11:


terrillmoore, 2023-08-11:
That's not really surprising -- the bootloader talks to external hardware. We already have two different bootloaders for hardware differences. Unless you're using the wrong bootloader (and the other one will work), this issue needs to be transferred to the bootloader. Careful hardware review is needed. You will probably want to single step through the bootloader with a debugger on the 4917 and make sure that every GPIO use and SPI use is correct. Probably there are power control things that need to be done.

dhineshkumarmcci, 2023-08-16:
thank you @terrillmoore . 

Yes, got to look at the power control of 4917 for Flash and it does needed 4801 Bootloader file, and it is tested successfully

closed 2026-09-24 by terrillmoore